0xPIT/encoder #7: Wiring Schematic

Nick1787, 2015-11-23:
Not really a code issue I apologize, but can you provide a basic wiring schematic?  I have your code up and running but it doesnt seem to be working properly in getting values of the encoder.  Im wondering if I have it wired incorrectly.

I realize it can be used on any pin, Im wondering if I have the wrong pins on the encoder going to ground or something.

Thanks

mrmonteith, 2016-01-11:
I agree.  At least say in the example which piece goes to which pin.   I mean the example shows A0, A1, and A2.  But which is wired to Clk, DT, + or what?   Thanks

malleYay, 2016-02-22:
Hey, the ClickEncoder.h says

`ClickEncoder(uint8_t A, uint8_t B, uint8_t BTN = -1, 
               uint8_t stepsPerNotch = 1, bool active = LOW);`

so the call 

`encoder = new ClickEncoder(A1, A0, A2);`

With a 5 (3 wire encoder, 2 wire button) wire encoder you should wire as follows:
Encoder A = A1
Encoder B = A0
Encoder Common = GND
Encoder Button_1 = A2
Encoder Button_2 = GND

Nick1787, 2016-02-22:
Awesome thanks!

- Nick

pille, 2018-01-22:
also solves #6