0xPIT/encoder #11: clear encoder value

mesmerise, 2016-03-09:
First of all: Thanks for this great library, it works just perfect! Really like the acceleration feature!

I wonder about one thing though: I use one click-encoder to navigate through a menu in which I set different values. By clicking the encoder I jump to the next setting, which can be changed by rotating the encoder. When I jump to the next setting, the value (based on the return value from encoder->getValue()) from the previous setting appears. I wonder if there is a way to reset/zero the encoder value or to define a preset?

Cheers,
Andy

soligen2010, 2016-03-09:
I believe getValue returns a delta, not the absolute value, so I think you want to keep track of of your setting in your own variable, and just apply the getValue delta to your variable.

If you look in the example program you will see the getValue is added to your own value:

value += encoder->getValue();

value is your own variable that you have control of, and you can reset it if you like.

soligen2010, 2016-03-09:
By the way, if there is something new that you want to contribute to the library, looks like 0xPIT isn't maintaining this, so you are welcome to offer pull requests for any updates to my fork for merge consideration.  You might want to get my fork regardless, as it has several bug fixes in it.

mesmerise, 2016-03-10:
Thanks for the quick reply! I was under the impression that value is a return value that cannot be altered. I just checked it and assigned additional variables without any trouble. In my defense, I'm a noob in C,  I'm used to programming in assembler..

Anyway, I will try your fork asap and try to contribute! Cheers!

closed 2016-03-10 by mesmerise