0xPIT/encoder #13: Arduino DUE ARM Platform

DanieGieb, 2016-06-18:
Hello,

i tried to compile the clickEncoder Lib for Arduino DUE Arm Board.
Unfortunatelly, i get some compile errors.

This is the log:
`C:\Users\Daniel\Documents\Arduino\libraries\encoder-master/ClickEncoder.h:18:20: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>`

Is there any sollution for this? I think avr.io is the wrong define for the registers of the DUE?

Thank you!
Regards Danie

soligen2010, 2016-06-22:
You could try simply removing the AVR includes.  This is what I did for the ESP8266 and it works. You can get my fork to see the #IF to remove the includes for the ESP.  If you get a new #IF worked out that works for DUE as well, you can submit a pull request to my fork - this original fork isn't being maintained.  

I also recommend to switch to my fork as I have fixed a number of bugs.

soligen2010, 2016-06-23:
I fixed the conditional in the soligen2010 fork for the AVR includes but it it failed with other errors (sei() and cli() not defined) so there are obviously other issues. I have never used a Due, so I cant really do any more.   If you get it figured out, please submit pull requests to the soligen2010 fork.

soligen2010, 2016-06-24:
The soligen2010 fork now compiles on the Due.  I don't have a Due so I cant test it, but I think it will work.

DanieGieb, 2016-06-25:
Hello, 

I have tried it. But unfortunatelly, i get an Compile Error!
`
exit status 1
'class TimerOne' has no member named 'initialize'
`
After installing the "TimerDue" libary from here: [https://github.com/ivanseidel/DueTimer](url)
and change line 48 to 49 from:
 ` Timer1.initialize(1000);
   Timer1.attachInterrupt(timerIsr); `
to:
`Timer1.attachInterrupt(timerIsr).start(1000);`

it compiles without any errors!

Now i am going to verify it with the Due as Hardware.
I will post my results here later!

Thank you for your support!
Regards Daniel

soligen2010, 2016-06-26:
I would be happy to accept a pull request for an update to the example. Also as a note, the ESP8266 example compiles for the Due - it works without a timer and works fine as long as the loop() function is not too slow.

Doc1012, 2017-04-27:
I also tried to work with ClickEncoder on the DUE. It worked perfect on the Mega. 
With the DUE I get an error when compiling:

`(x86)\Arduino\libraries\encoder-master/ClickEncoder.h:52:11: error: 'WiringPinMode' does not name a type
   typedef WiringPinMode pinMode_t;
           ^`

I found this in the ClickEncoder.h:

`#if    defined(__arm__)
  typedef WiringPinMode pinMode_t;
 #else
  typedef uint8_t pinMode_t;
#endif`

It seems, that the WiringPinMode is somhow missing?

What can I do?

soligen2010, 2017-04-28:
What I think is happening is that the STM23duino and Due are both arm chips, but need different typedefs.  This line was added to support STM32duino, but must have broke the Due becasue it needs the second typedef.  What we need is another conditional compile #if specifically for the Due.  If you can find out what the pre-compiler variable is for the Due, should be easy to change the #if.

Doc1012, 2017-04-28:
After a Little Research I found:

For all Arduino Due compatible boards:
`#if defined (__arm__) && defined (__SAM3X8E__) // Arduino Due compatible
// your Arduino Due compatible code here
#endif`

(found in file sam3.h. More info there.)

If you just want to define Due (by leaving out compatible boards):

`#if defined (_VARIANT_ARDUINO_DUE_X_)
// your Arduino Due code here
#endif`

(It's defined in variant.h file of Arduino Due)

I hope, this is what you wanted.

soligen2010, 2017-04-29:
just uploaded a fix.  Please try it out

Doc1012, 2017-04-29:
Sorry for that stupid question:

I can't find a new/updated file. Where is it?

soligen2010, 2017-04-30:
Its the soligen2010 fork.  I was assuming that is what you are using.  As  0xpit doesn't pay much attention to this library, there is no way to update the original

Doc1012, 2017-05-01:
Now it's clear....

It works! Your trick did the job. Thank you!

HFATL, 2018-10-06:
Hello,

I'm also getting: 'class TimerOne' has no member named 'initialize'
I need timer interrupts on my ESP8266.

Please help.

Thank you