0xPIT/encoder #27: Is each indent supposed to move counter by one?

moballa, 2018-01-13:
I have a rotatory encoder with pushbutton.  I am using this library to read the values from the encoder.  Each time I move the encoder one indent, 4 events get triggered advancing the counter by 4.  Not sure if this is normal or my encoder is different than the usual expected behavior.  I have tried adding #define ENC_DECODER (1 << 2) before including the .h file.  I have also tried #define ENC_HALFSTEP with different values with 2 or 3 events getting triggered instead.  When I set the 4th parameter of the constructor to 2, only 2 events get triggered but if I change it to the default of 1, 3 events get triggered.

Is there a problem with my encoder?  Is there a way to ensure that only one event is triggered per encoder indent so that it only advances the position counter by 1 per indent?

Thanks for any help!

CyrilSiman, 2018-03-04:
Hi, 
I had the same problem on my side and maybe could you fix it with stepsPerNotch in constructor as mentioned in doc : 

> Depending on the type of your encoder, you can define use the constructors parameter stepsPerNotch an set it to either 1, 2 or 4 steps per notch, with 1 being the default.

On my side with a Keyes encoder, I have to specify 4.

floony, 2019-01-31:
Im not very experienced with libraries for arduino, so could someone please explain how to define the stepsPerNotch?
Thanks a Lot!

themodfatherinc, 2019-03-28:
i just had to figure this out, as mine sent 2 per ident. The number of stepsPerNotch is defined as a 4th variable after the analog pins as below.. In this example it defines 2 steps per notch

void setup() {
  Serial.begin(9600);
  encoder = new ClickEncoder(A1, A2, A3, 2)

The referenced part in ClickEncoder.h
  ClickEncoder(uint8_t A, uint8_t B, uint8_t BTN = -1, 
               uint8_t stepsPerNotch = 1, bool active = LOW);

chopsuwe, 2019-09-25:
I have the same problem. My encoder has 8 steps per notch. I have tried all the methods listed above, both setting the 

stepsPerNotch = 8 

as well as setting 

stepsPerNotch = 4    ... and ...
#define ENC_DECODER (1 << 2) 

which I hoped would divide by 4 then by 2 resulting in 1. But I can only get down to 2 steps per notch. Can anyone help?